viesturz/tapchanger #1: Canbus setup and wiring

sgermain06, 2023-05-24:
Hi! Thank you for such an amazing project! I'm really looking forward to being done printing it and start installing it on my Voron 2.4! I just have a question regarding the canbus wiring. Which main board are you using and how did you wire up your canbus to support multiple printheads?

Thanks!

AugieSpear, 2023-05-24:
Hi, I have setup the fly sb2040 canbus board and used the fly u2c to connect it directly to the rpi. Im pretty sure you can daisy chain the data signals and end off the chain with a 120 ohm resistor (usually a jumper on the canbus board).

sgermain06, 2023-05-24:
Ah, that makes sense. That means, I could split off the positive and
negative, but connect D+ and D- between the multiple devices in serial,
like RPI -> device 1 D+, device 1 D- -> device 2 D+, device 2 D- -> RPI?

viesturz, 2023-05-24:
Can bus is a bus, so all d+ and d- are connected in parallel along the main bus wire.
The toolchanger setup is essentially a star formation, which is poorly supports by Canbus.
But runs fine on 500kbs speed and getting around 10% bus utilization with 3 toolheads.
The Canbus specs has wire length limitations depending on configured speed. 
https://community.nxp.com/t5/LPC-Microcontrollers/CAN-FD-wire-length/m-p/1063937
For star layout look at max stub length and accumulated stub length.

I have all the hotend wires connected in parallel inside the filter box at the back and then a beefy power cable going down to PSU. And 120 ohm terminators at USB to can adapter down at the base plus another one in tool 0.

I'm using a canable USB to can board, but have had good success with the Chinese boards as well.
Big tree tech M8P built-in USB to can was unreliable for me though. I suspect mainly because it makes the main board a can node as well and has a lot of traffic to drive all the motors and fans.

sgermain06, 2023-05-24:
Yeah, I'm planning on using the RJ-11 on the BTT Octopus board. Thanks for the information!

closed 2023-06-28 by viesturz