viesturz/tapchanger #2: Discussion on some issues encountered during the build

EiNSTeiN-, 2023-05-28:
Hi! I just had my first successful load/unload, so first of all thank you for this amazing project!

I'm opening this github issue to discuss some of the pain points I had while they are fresh in my memory hoping this information may be useful to someone else. I'm more than willing to open pull requests to add documentation or modify parts to fix these issues, but I wanted to get some eyes on them first, in case any of them are due to me doing something wrong!

1) I had some parts break due to printing in the wrong orientation, for example the strain relief that goes on the StealthBurner. It would be useful to get them oriented correctly when exporting!
2) Front plate: The holes for the the inserts under the rail are not long enough, the plastic pools at the back and pushes against the inserts, preventing them from sitting flush with the top of the plastic part.
3) Tap PCB: The sensing zone on my optical sensor was too far back to sense the shuttle arm being inserted. I had to add two washers between the PCB and the plastic part to move the sensor forward a bit. Unfortunately I got the PCB from another user on the voron discord so no idea of the origin, whether it's just this one model or a widespread issue.
4) Shuttle: I got pan head screws from McMaster, they fit in the shuttle holes very snug and were impossible to actually turn. I ended up drilling the holes with a flat endmill but some additional clearance would be good.
5) Shuttle: Making it 3mm thicker to recess the carriage screws more would make it possible to use BHCS screws with a ball end allen key, I think that would be more convenient than the flat end screws! I got a box of 100 but don't see myself using the rest of them for anything... With the added extrusion adding 20mm at the front of the printer, I think some added thickness to the shuttle would work just fine.
6) Belts: I originally tie-wrapped the belts but didn't insert them in the little plastic loops on the side of the shuttle, but that causes interference while loading/unloading. I suspect I'm not the only one who will cut corners on this...
7) Dock: I broke the front part where screws catch on the magnets while installing! It needs ridges at 90 degrees at the back to make that bit stronger.
8) Dock: I used 3x18mm dowel pins which removes the need to use a dremel and modify some screws. I used a 3mm reamer on the dock for friction fit and a 1/8" drill on the hinged part for a loose fit. I think this step will be a source of variability from build to build so using an off the shelf part instead of a hand-made part would be really nice.
9) Dock: Broke multiple of them while inserting the magnet in the hinge, a little bit more clearance or a break tab designed into the part would be great.
10) How do you hold the magnets at the front of the SB? The 5mm magnets are too small, and the 6mm ones are too big. I ended up reaming the holes to 6mm and inserted a magnet with a really good friction fit. However it'll be impossible to remove them without breaking the SB cover :( For now I left the magnets out and I can load/unload the tool just fine. but I can see the magnets would help stabilize the toolhead.
11) The config seems to break the regular behavior of M104/M109, the commands just don't have any effect unless there is an active tool, which isn't the case unless `pickup_gcode` runs once (the `M568` line is what sets the active tool), so right after starting the printer there is no active tool and temperature gcodes don't have any effect. I lost a bit of time figuring that one out, I think adding `M568 P{printer.tool_probe_endstop.active_tool_number} A2` to `check_active_tool_on_startup` might be an appropriate fix for this?

Other small details worth adding to the documentation somewhere
* Would be great to specify the location of the extrusion onto which the docks are attached in relation to your printer frame. For example I attached mine with 150mm spacing between the two extrusions, and the first dock is attached 30mm to the left of the frame. I mention this because I crashed my gantry due to wrong dock positioning!
* `position_min` must be adjusted for Y to `-10` to reach the dock, and must make sure the gantry allows for that movement!
* `position_max` should be adjusted for Z to 340, I believe the default is 310 (that's what mine was set to)

viesturz, 2023-05-28:
Thanks for this!
A lot of super useful info, will work though them one by one.

It would be useful of you can split this out into multiple bugs - general
printed parts fit; design ideas; configuration, etc.

Also some pictures would be very appreciated.

One note: I'm printing all this stuff with outer layer first which leads to
better tolerances and slight bias towards undersized parts.

Some initial replies:

1: any good tools to do that? Fusion360 seems overkill to
load-rotate-export.

2: yes this is a tradeoff between extra Y distance and installation
convenience. I'm just trimming off the extra with a hobby knife. Will
experiment with this some more.

3: my PCBs work just fine. The pcb alignment is exactly the same as in
regular Tap. Suspecting your part is off.

4: can you please post the diameter of your screws?

5: cool idea! Can you make a test jig and figure out how many mm is a good
distance?
I might get around to it at some point, but that would speed things up.

6: yeah good point, updating the instructions.

7: never have broken it. But I did made it thinner on last update to add a
bit more flex for more Y tolerance.

8,9: I'm considering of the pivoting part is useful at all. Might be more a
finicky distraction than good.

10: I use black steel bolts and 5mm magnets - they are just magnetically
sticking to the bolt heads.

11: if you can describe your SW setup a bit more?
The gcode macros are copied from TypQxQ repo. Plus an extra delayed macro
to detect the currently mounted tool at boot time. Maybe that one is not
working for you?

Cheers!
Viesturs

EiNSTeiN-, 2023-05-30:
I'll split this issue up into new ones :+1:

EiNSTeiN-, 2023-06-10:
I think all these are addressed

closed 2023-06-10 by EiNSTeiN-