viesturz/tapchanger #13: proprietary cad format

sirus20x6, 2023-09-27:
can we get the cad files as step as well? it would be nice to be able to use freecad to open the files and work on them

viesturz, 2023-09-27:
Hi, will add step files later today.

viesturz, 2023-09-27:
Added step files for plates and square rods, the round rods version needs a bit more work - I modified it to make square rods, will get back with that soon.

alfskaar, 2023-10-23:
Can we have step files please.

viesturz, 2023-10-23:
Added, let me know if this works, unfortunately the square and round rods are the same model. I'm looking for a way to add both to the same assembly without too much hacks. Let me know if you need the round rods version.

alfskaar, 2023-10-24:
Thanks. Yes I where thinking absolutely using the "round rods" I have printed the two may parts to get a feel for this. I wondered if the end of the M5 screw can be used. I will have close look and see If can find something that is more common available in my tools.

alfskaar, 2023-10-26:
I wonder it whole have be easier if you made several STEP files as I’m  not really sure with one. ? For AFTERBURNER or Stealthburner

[image]

viesturz, 2024-02-19:
The cad now has no holes at all, but includes the hole shapes - substract the one you need.

closed 2024-02-19 by viesturz