viesturz/tapchanger #26: JST-XH connector used on CAN distribution board only supports 3A

esb7, 2024-01-25:
If I'm reading the connector specs right (and reading the right connector specs), the JST-XH connector only supports up to 3A.  For a 24v hotend, that maxes out at 72W.  Ideally you'd derate it a bit, so realistically you'd only want to support ~60W or so.  And that power also needs to drive the toolhead PCB, stepper driver, fans, etc... So the largest hot end you'd likely want to run on that is a good bit less than 60W.  It would depend on the specs of the various components being used.

It may be better to use something like the 4-pin Molex Micro-Fit 3.0mm connector that is used on the U2C and EBB36 (I think).  It appears that supports up to 8.5A which should be plenty for the largest of hotends.

M4dMike, 2024-01-25:
+1 had the exactly same thought.
There are even "advanced" Microfit 3 inserts that support up to 10A.

esb7, 2024-01-26:
I should also add that the connector / wiring that connects back to the power supply needs to support the full load of power if all of the hot ends happen to be heating at once.  In this case, that's 6 x ~100W+ == 600W+ @ 24v == 25A.  The 12awg wire used in the example is probably the right size to use.  For the connector, maybe use JST-XH for the two canbus conductors and then a pcb mount terminal block that can handle 25A+ for the supply power lines.

esb7, 2024-01-27:
One thing that was pointed out to me is that typically a hotend / heater will only pull max power during the initial heat up phase.  Once at temperature, it takes less power to keep it heated.  So if the heat up process for each hot end is staggered during the initial print start routine, the total power needs can be lowered.  However I don't have a good estimate for what the power requirements would be in that case.  Maybe something like a 50% - 100% increase over max power for a single toolhead would be sufficient.

viesturz, 2024-02-10:
Thanks for all the commets. I agree the board is not suitable here. Moving to a Old stuff folder.
Personally I'm using wago clones: https://github.com/viesturz/tapchanger/blob/main/images/umbilical-plate.jpg

closed 2024-02-10 by viesturz