viesturz/tapchanger #27: Stealthburner Toolhead support for TapSensor is too weak

orotecluc, 2024-02-19:
Hi,
Iam trying the StealthBurner Square Rods version of  Tapchanger/STL/Toolheads/Stealthburner.
But everytime, the mount to where Ihave to put the Tap Sensor, breaks off.
Since I have no Fusion360 experience at all, wouldn't it be possible to give it a more sturdy top wall, in line with the mount on the Voron-Tap? 
I know, on Voron Tap, the stup is completly different, with a seperate peace to mount the sensor, and heavy side walls on both sides of the sensor, to make the mount more sturdy, but I think, with a thicker top layer on mount piece, we can have already a good result.

viesturz, 2024-02-19:
Can you add a picture how exactly it breaks off?
Does it break during assembly? Or when printing?

orotecluc, 2024-02-19:
It breaks OFF during assembly.

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

orotecluc, 2024-02-19:
[image]

The bridge detaches rally easaly, but i thank, if we kake the backpart of that bridge stronger, Luke in my tinkercatd try, everything Will keep hold

viesturz, 2024-02-19:
Thanks, making a more robust version...

viesturz, 2024-02-19:
There, added 2 mm of plastic, let me know if this works.

closed 2024-02-19 by viesturz

orotecluc, 2024-02-19:
sorry, late out this evening. Testprinting right now.
Thanks for the fast response

orotecluc, 2024-02-20:
Prints are great now, mounted complete Square-rod-toolhead. Just had some issues mounting the TAP sensor, BHCS6mm are a bit too long, it seems, and screw holes are really narrow. Double magnets are a great upgrade.