viesturz/tapchanger #36: OMG... the new DB Cowling design.

walterwissmann, 2024-03-11:
The old one with the markers was a better design and his functionality was very good ;-)

viesturz, 2024-03-11:
Thanks for the input. I agree the new ones are a bit fugly.
Moved away from main branch for now.

So there are two cahnges I'm trying to do with this:
- move the hook points further apart for better docked stability.
- allow for ~1mm tolerance in all directions.

Let me know if you have some ideas how to better get there.

viesturz, 2024-03-11:
I sketched up an alternative. Kinda a middle ground between all form and all function.
Will test print and upload. 
Let me know if you have ideas.

[image]

walterwissmann, 2024-03-11:
The look and feel of the old version with the markers has a high-quality and friendly appearance:

[image]

Two thinks i have done per hand to have a super functionality: 

a. i have make the front holes a little bit grater, 6,5mm.

b. I cut the plastic spacers for the suspension bolts on the park mount a little conically so that the plastic remains behind the bolt and the Dragonburner doesn't bump into it.

This allows the cowling to slide easily over the screw into position. 

[image]

BR
roerich_64 

p.s. we can discus via discord. i am also there ;-)

closed 2024-04-03 by viesturz