viesturz/tapchanger #43: Beacon Support

gfacek, 2024-05-08:
I just finished my Voron 2.4 a few weeks ago, and since I decided to build it, I've been following this project because I was hoping to replace my old IDEX with it.

Since the Tap Kit from Chaoticlab broke after a short time, and I had been toying with the idea of a Beacon anyway, I swapped the Tap for Beacon (since Beacon can't do CAN, I use Cartographer 3D) and am simply thrilled with how reliably and quickly it works.

So much for the preface.

I'd like to tackle this tool changer next, but I'm wondering how best to implement it with Beacon. I don't think it makes much sense to build a beacon for every toolhead. Hence my thought:
Does it perhaps make sense to add a mount for the beacon to the shuttle? You would then have to pick up each toolhead to be used once at the start of printing in order to calibrate Z.

Would something like that work?

viesturz, 2024-05-20:
I would probably go with adding a beacon to a single toolhead.

Or alternatively add more stuff to shuttle.
I have considered moving the part cooling fans on the shuttle, but the space is really tight.

colin342, 2024-07-08:
Hi

I did setup two tools and one tool has the cartographer3d the other has tap. Now I am struggling to setup the probes part. Once the cartogrpaher is setup and I import the tool_detection.cf, I get the error Duplicate chip name 'probe'. The thing is I kinda understand why, but I have no idea if it can be solved.

gfacek, 2024-07-13:
I'm still waiting for my parts I ordered. But I also think, this is the best solution. So I can "only" have 5 tool for printing, but 1 tool for bed mesh with cartographer.

Do you document your your project anywhere? I'm also interested in helping to design. I'm not very experienced, but can somehow handle with fusion 360 (a bit) 😅

nemesit, 2025-02-03:
ideally the beacon probe could be on the shuttle so all heads could use it.

Benkult, 2026-10-03:
https://www.printables.com/model/1496221-rossinante-tap-shuttle-mount-for-tapchanger