viesturz/tapchanger #44: Shuttle issue below belts

vikwin, 2024-05-15:
I noticed that the non-aluminum shuttle warps a lot when it is attached to the linear rail. When looking for the cause, I saw that the smooth surfaces next to the belts are not in the same plane, but are offset. 
I noticed this with the “old” version from before 2024-05-08, but it is also the case with the newer version.
[image]

viesturz, 2024-05-16:
Hmm, indeed. Let me try to fix it.
The belt groove profile was imported from previous iteration, perhaps
something is messed up there.

This aside, there should be some deformation from bolting down the shuttle
from the belt compression. You should be able to compensate by adjusting
the bottom screws on each toolhead.

closed 2024-05-16 by viesturz

vikwin, 2024-05-17:
Thanks for your fast reaction, I really appreciate your efforts for the community with tapchanger!
Do you have any plans on removing the central area padding for the aluminum version as well?

viesturz, 2024-05-17:
I added separate keeper optimized versions instead.

commit 764a14c1 (2024-05-16): Fix shuttle base misalignment.

Fixes #44